vdudouyt/stm8flash #12: Can't program flash protected by ROP

eddyem, 2015-02-03:
stm8flash can't clear flash of STM8S003 protected from reading by ROP:

stm8flash -cstlinkv2 -pstm8s003 -w testproj.bin
Determine FLASH area
Writing binary file 2271 bytes at 0x8000... Tries exceeded

The same problem when I try to clear ROP:
stm8flash -cstlinkv2 -pstm8s003 -s 0x4800 -b 1 -w ROP_clear.bin 
Writing at 0x4800... Tries exceeded

TorstenRobitzki, 2015-06-26:
I see the same error message, when trying to flash a STM8L051F3. I would try to fix this, if someone could provide me some documentation to st-link/v2.

baruch, 2015-06-26:
ST micro have all the docs on their website.

Baruch

TorstenRobitzki, 2015-06-26:
Hi @baruch,
please forgive my ignorance, but could you provide me a link to the st-link/v2 protocol specification? I have problems finding them.

cheers,
Torsten

TorstenRobitzki, 2015-06-26:
Anyway, I got the error message wrong. I was able to add support for my device by simply adding a new device type.

spth, 2020-05-20:
Are there any changes you made that should be comitted to the master branch? If yes, how to test them?

TorstenRobitzki, 2020-05-20:
Sorry, can't remember any details. Tooooo long ago :-)

spth, 2020-05-20:
Closing: Out of date.

closed 2020-05-20 by spth

mmilch, 2020-06-11:
_$ echo "00" | xxd -r -p >ROP_CLEAR.bin
$ stm8flash -c stlinkv2 -p stm8s105?4 -s opt -w ROP_CLEAR.bin_ 

Hi. I got ROP protected stm8s105s4 chip in e-bike controller and would like dump it's entire content (flash, eeprom and option memory). Is it 100% sure that doing above trick will result in a lost of original flash memory ?