vdudouyt/stm8flash #15: cannot compile under MacOS X

gicking, 2015-04-19:
hi Valentin,

I would very much like to use your stm8flash tool under MacOS X 10.10.3. However, I ran into a problem for which I need your help. Here's my steps with observations:

* installed pkg-config and libusb --> worked

* compile of stlink.c gave 

    * many warning messages like (ignored):
    *stlink.c:37:71: warning: passing 'char [32]' to parameter of type 'unsigned char *' converts between pointers to integer types with different sign [-Wpointer-sign]*
    --> suppose this is not critical...? Disabled all warnings by *CFLAGS += -w* in Makefile 

    * many error messages like (fixed):
    */usr/local/Cellar/libusb/1.0.19/include/libusb-1.0/libusb.h:1696:10: error: use of undeclared identifier 'NULL'*
    --> fixed by *#include \<stddef.h>* in stlink.c before *#include \<libusb.h>*

    * many error message like (open):
    *stlink.c:340:12: error: use of undeclared identifier 'stderr'*
    *fprintf(stderr, "Unknown status: %x\n", status);*
    In addition it warns that 
    *declaration of built-in function 'fprintf' requires inclusion of the header \<stdio.h>*
    --> it seems like the inclusion of *\<stdio.h>* in line 4 of stlink.c has no effect...  :-(

* to make sure my toolchain is ok, I compiled a dummy program with *fprintf(stderr,"test");* and it was ok

--> I played around a lot but can't get that stdio.h point fixed :-(  Any idea what's going on here? For your help thanks a lot in advance!

Regards, Georg

vdudouyt, 2015-04-20:
Hi Georg,

This should be fixed by  #fbfd745 (thanks to Philipp). Please confirm.

Best Regards,
Valentin

gicking, 2015-04-25:
hi Valentin,

unfortunately there seem to be different, incompatible versions of C compilers on OS X :-1:     In any case, after correcting some header files I got it to compiler on my Mac without further warnings or errors. However, when I try to flash a hexfile to a STM8 discovery board, I get the following error message:

Determine FLASH area
libusb: warning [ep_to_pipeRef] no pipeRef found with endpoint address 0x02.
libusb: error [submit_bulk_transfer] endpoint not found on any open interface
Assertion failed: (actual == USB_CBW_SIZE), function stlink_send_cbw, file stlink.c, line 162.
Abort trap: 6

The setup is:
 - discovery board directly connected to Mac (no USB hub)
 - in system profiler it shows up as STM32 STLink (product-ID: 0x3744, vendor-ID: 0x0483) 

Any idea what's wrong here? Sorry to keep bothering you with this, but as far as I can see, there is no alternative to stm8flash, except Windoze...

Any help is highly appreciated!
Georg

gicking, 2015-05-05:
hello all,

issue solved (thanks Tim!). See [here](https://github.com/vdudouyt/stm8flash/pull/16) 

Georg

closed 2015-05-05 by gicking