vdudouyt/stm8flash #24: compile fails under MacOSX 10.11.1 "el Capitan"

gicking, 2015-10-31:
hi Valentin,

after updating to MacOSX 10.11.1 "el Capitan" I wanted to check if claiming the USB port now works under OS X (see also https://github.com/vdudouyt/stm8flash/pull/16). Unfortunately make aborts with the below errors. Any ideas? Thanks a lot in advance!

$ make
cc `pkg-config --cflags libusb-1.0` -g -O0 --std=gnu99 --pedantic -I`xcrun --show-sdk-path`/usr/include/ -I`xcrun --show-sdk-path`/usr/include/sys -I`xcrun --show-sdk-path`/usr/include/machine   -c -o stlink.o stlink.c
stlink.c:247:3: warning: implicit declaration of function 'usleep' is invalid in C99 [-Wimplicit-function-declaration]
                usleep(10000);
                ^
stlink.c:342:4: warning: implicit declaration of function 'fprintf' is invalid in C99 [-Wimplicit-function-declaration]
                        fprintf(stderr, "Unknown status: %x\n", status);
                        ^
stlink.c:342:4: warning: declaration of built-in function 'fprintf' requires inclusion of the header <stdio.h>
      [-Wbuiltin-requires-header]
stlink.c:342:12: error: use of undeclared identifier 'stderr'
                        fprintf(stderr, "Unknown status: %x\n", status);
                                ^
stlink.c:542:12: error: use of undeclared identifier 'stderr'
                        fprintf(stderr, "Write error\n");
                                ^
3 warnings and 2 errors generated.
make: *** [stlink.o] Error 1

Best regards,
Georg

njzhangyifei, 2015-11-16:
Installing Xcode command line tool by running`xcode-select --install` from command line will fix this problem.

Tested on Mac OS X Yosemite 10.10.3. I can confirm this fixes the problem and I can use `stm8-flash` to program the chip.

gicking, 2015-11-17:
hi njzhangyifei,

yep, that solved it under MacOS X 10.11.1. Thanks a lot!  :-)

Georg

closed 2015-11-17 by gicking