vdudouyt/stm8flash #26: stm8flash not work on archlinux raspberrypi

idreamerhx, 2015-12-08:
Hi,

uname -a:
Linux betta 4.1.13-1-ARCH #1 PREEMPT Fri Nov 13 20:10:41 MST 2015 armv6l GNU/Linux

stm8flash is the latest master code.
I just clone from git and make.

The command line I call stm8flash in Makefile is:
stm8flash -c stlinkv2 -p stm8s103 -w blinky.ihx uart.ihx

This not work. Seems that main.c (line: while((c = getopt (argc, argv, "r:w:v:nc:p:s:b:")) != -1)) getopt returns wrong result.
If I add "if(c == 255) break;" to first line in the while block, It can break and execute following code.

idreamerhx, 2016-01-08:
I pull the latest  code. It works different on archlinux (raspberrypi) and ubuntu14 (x86).

On archlinux  it print "Usage" and exit.

On ubuntu14 it print assert.

fragrans, 2016-05-07:
I had the same problem. and I had to break out of loop when it met 255 as a workaround.
Then, I found this link. and it works like a charm.
http://stackoverflow.com/questions/17070958/c-why-does-getopt-return-255-on-linux

spth, 2017-07-03:
This seems to be fixed by an explicit cast of (-1) to char in the current version. Can you check if it works for you, so this issue can be closed?

Philipp

spth, 2018-01-19:
A fix has been committed half a year ago. Since no one reported having this issue since, I assume it works.

Philipp

closed 2018-01-19 by spth