vdudouyt/stm8flash #27: STM8L150G4 - clearing ROP option

lwojdziak, 2015-12-14:
Hello, 

I have problem with clearing ROP (Read-Out Ptotection) byte in STM8L150. 
I have device with protected bytes set.
I trying to set this byte to 0xAA, then reset the device and again set ROP to 0xAA - according ST PM0054, p.5.2.
After this procedure ROP is still equal 0x00 and writing is disabled.

I checked that I can freely modify bytes 0x4801 - 0x480C. 
I can't modify only 0x4800.

BR

spth, 2015-12-28:
What's STM8L150G4? An STM8 variant no longer or not yet available? Google finds only this ticket.

Philipp

lwojdziak, 2016-01-04:
Sorry, my fault. Of course this is STM8L151G4U6.

veinmichal, 2016-06-14:
Hello, 
Has anyone solved this issue? According to ST Programming manual:
"To unprotect the device, the ROP must be written twice: the first write, with any value, 
launches a global erase which includes the option byte; the second write, 0xAA, 
unprotects the device, and then the 0xAA value remains in the ROP byte. 
Please note the EOP should be checked each time. "

However launching the procedure twice doesn't make it even though it says 'Writing  binary file 1 bytes at 0x4800... OK'.

spth, 2016-06-15:
The branch by cpldcpu could have a solution.

Philipp

lkord, 2018-02-25:
Issue still unsolved on stm8l151g6. I manged to unlock rop with openocd but stm8flash still can not do this.

lkord, 2018-02-27:
Today I tried writing 13B: AA 00 00 ..... inside-u(nlock) parameter implemented:
write 0xAA
read eop -> 0x14 (EOP bit is 1)
write 0xAA
I put it in UNLOCK action after in block of if (part->read_out_protection_mode == ROP_STM8L){}

if (part->read_out_protection_mode == ROP_STM8L) {
			fprintf(stderr, "ROP_STM8L\n");
			
			int bytes_count_align = ((bytes_count-1)/256+1)*256; // Reading should be done in blocks of 256 bytes
			unsigned char *bufr = malloc(bytes_count_align);
			if(!bufr) spawn_error("malloc failed");
			
			
			for (int i=0; i<bytes_to_write;i++) {
				buf[i]=0xAA;
				if (i>0) buf[i]=0x00;
				fprintf(stderr, "%x ", buf[i]);
			}
			/* flashing MCU */
			sent = pgm->write_range(pgm, part, buf, start, bytes_to_write, memtype);
			
			int recv = pgm->read_range(pgm, part, bufr, 0x5054, bytes_count_align);
			for(int b=0; i<sizeof(bufr);i++){
				fprintf(stderr, "\nBytes read:");
				fprintf(stderr, "%x ", bufr[i]);
			}
						
			if(recv < bytes_count_align) {
				fprintf(stderr, "\r\nRequested %d bytes but received only %d.\r\n", bytes_count_align, recv);
				spawn_error("Failed to read MCU");
			}
			
			sent = pgm->write_range(pgm, part, buf, start, bytes_to_write, memtype);
		}
but thi still doesnot work.

lkord, 2018-04-24:
for future questions: https://github.com/vdudouyt/stm8flash/pull/98
Issue Solved.

spth, 2024-12-24:
AFAIK, we have had working ROP support for quite a while.

closed 2024-12-24 by spth