vdudouyt/stm8flash #33: Release it

spth, 2015-12-26:
stm8flash currently is an essential part of the free toolchain for STM8. I'd suggest to make a release. Releases tend to be more trusted by users and tend to make it more likely that the software gets picked up by distributions. There is probably still a bit of work in getting stm8flash into a state fit for release (and I'm willing to help), but even in its current state it is quite useful software that should be good enough for the vast majority of use cases.

Philipp

spth, 2020-08-05:
Also, a release could come with Windows binaries.

Since building software for Windows is much harder than for other OSes, this would be very helpful for Windows users.

schneidersoft, 2023-04-21:
Under windows I compiled stm8flash using mingw once. I had to remove support for the esp uart bootloader stuff to get it to build though.