vdudouyt/stm8flash #34: Change license from GPLv2 to GPLv2 or later.

spth, 2015-12-26:
The README.md states "This is a free and opensource software distributed under the terms of GNU General Public License v2."

I suggest to change the license to GPLv2 or later, i.e. change the text to ""This is a free and opensource software distributed under the terms of GNU General Public License; either version 2 of the License, or (at your option) any later version."

This would be a license change, so all authors would have to agree.

However, I think it would be good to make this change, to allow combining stm8flash with software released under later GPL versions, should the need arise. It would be quite frustrating to not be able to do some things due to license issues, even when all involved software is GPL-licensed.

Philipp

rpavlik, 2017-03-24:
I've only made minor contributions, but please feel free to relicense my contributions as GPL2+.

cpldcpu, 2019-02-07:
My contributions are minimal, no objection from my side.

spth, 2021-09-26:
Done.

closed 2021-09-26 by spth