vdudouyt/stm8flash #35: problem flashing 8S003F3 with stlinkv2

kensenjiha, 2015-12-29:
to flash a protected STM8S003F3, using single byte 0x55 file ROP_CLEAR.bin to erase ROP:

```
$ echo "aa" | xxd -r -p >ROP_CLEAR.bin
$ stm8flash -c stlinkv2 -p stm8s003f3 -s opt -w ROP_CLEAR.bin 
Determine OPT area
Writing binary file 2 bytes at 0x4800... OK
Bytes written: 1
$ stm8flash -c stlinkv2 -p stm8s003f3 -s flash -r nothing.bin ; xxd nothing.bin
Determine FLASH area
Reading 8192 bytes at 0x8000... OK
Bytes received: 8192
00000000: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000010: 0000 0000 0000 0000 0000 0000 0000 0000  ................
.
.
```

(very strange, erased flash should reads ffffff) Now the MCU really is not running old program, but it cannot be flashed

```
$ make flash 
stm8flash -cstlinkv2 -pstm8s003f3 -w blink.ihx
Determine FLASH area
Writing Intel hex file 202 bytes at 0x8000... Tries exceeded
```

erase it again

```
$ stm8flash -c stlinkv2 -p stm8s003f3 -s opt -w ROP_CLEAR.bin 
Determine OPT area
Writing binary file 1 bytes at 0x4800... OK
Bytes written: 1
$ stm8flash -c stlinkv2 -p stm8s003f3 -s flash -r nothing.bin ; xxd nothing.bin
Determine FLASH area
Reading 8192 bytes at 0x8000... OK
Bytes received: 8192
00000000: 7171 7171 7171 7171 7171 7171 7171 7171  qqqqqqqqqqqqqqqq
00000010: 7171 7171 7171 7171 7171 7171 7171 7171  qqqqqqqqqqqqqqqq
.
.
.
```

repeat erase and read, readout data stuck in this 000000 and 717171 loop.

same issue on ubuntu 15.10 and OS X 10.11.2

```
$ uname -a
Linux [hostname] 4.2.0-22-generic #27-Ubuntu SMP Thu Dec 17 22:57:08 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux
```

```
# uname -a
Darwin  [hostname] 15.0.0 Darwin Kernel Version 15.0.0: Sat Sep 19 15:53:46 PDT 2015; root:xnu-3247.10.11~1/RELEASE_X86_64 x86_64
```

kensenjiha, 2015-12-30:
additional info:

today i tried using STVP to read. STVP says this chip is protected. after erased using STVP(windoze in a virtualbox), stm8flash (both linux and osx) can flash the chip again.

guess we can narrow down to opt page programming.

lwojdziak, 2016-01-11:
Hello! You have to write opt space with two bytes at minimum: [0][1] = [byte][negation of byte].

kensenjiha, 2016-01-11:
oh, thanks, I didn't know that! 

but the ROP has only one byte. 

anyway i'll try later.

lwojdziak, 2016-01-11:
Yes, but each option byte in stm8s003 is saved as a pair: b and ^b in next cell.

kensenjiha, 2016-01-12:
oh, sorry~ I understood the RM0016 the wrong way. should write a byte other than 0xAA @0x4800 to disable ROP! 
Thanks~

closed 2016-01-12 by kensenjiha

mmilch, 2020-06-11:
_$ echo "00" | xxd -r -p >ROP_CLEAR.bin
$ stm8flash -c stlinkv2 -p stm8s105?4 -s opt -w ROP_CLEAR.bin_ 

Hi. I got ROP protected stm8s105s4 chip in e-bike controller and would like dump it's entire content (flash, eeprom and option memory). Is it 100% sure that doing above trick will result in a lost of original flash memory ?