vdudouyt/stm8flash #36: Writing outside of the lower 64K

spth, 2015-12-30:
I did a bit further testing on chips with more than 32K (results in the README.md of my branch). It seems I cannot write outside the lower 32K of flash, i.e. I cannot write outside the lower 64K of the address space.

Philipp

welash, 2016-05-20:
Out of curiosity, were you trying to write an address above 64k by using an intel hex file?  I just looked at the intel hex code (to add the ability to create intel hex files on output) and it doesn't look like the current code handles the extended segment or extended linear address records from intel hex files that allow them to handle addresses above 64k.  I don't think it would be hard to add these (basically just detect the record types and update an "offset" variable).

spth, 2016-05-23:
AFAIR, I was using a binary file. And looking at the stm8flash code, it seemed it can't handle addresses > 64K in the backends either.

Philipp

janezNov, 2017-06-05:
@vdudouyt 
What was the your source of USB protocol for STLinkV2.

It seems you are packing HI() and LO() parts of start address similar to:
[image]

Instinticly I tried adding #define EX(x) (((x) & 0xff0000) >> 16)

stlink2_cmd(pgm, 0xf40a, 7,
			0x00, 0x01,
			0x00, EX(start),
			HI(start), LO(start),
			byte);

Does not seem to work!
Thank you on the hints!!

janezNov, 2017-06-07:
I have started to capture USB packets.
Now need to compare it to @vdudouyt code.
I am flashing only one byte with ST Visual Programmer

1. need to understand writing at address 0xFFFF (all sequences before actual byte transfer)
2. comparing writing to 0xFFFF and 0x10000
3. sequence until writting

I append a "Spreadsheet"
[1byte_0x66_write_0xffff_0x10000_comparison.xlsx](https://github.com/vdudouyt/stm8flash/files/1057210/1byte_0x66_write_0xffff_0x10000_comparison.xlsx)

**Quick comparison:**
0xFFFF:
00 02 00 00
F4 0A 00 01 00 00 FF FF 66 FF 00 00 00 00 00 00

0x10000:
00 02 00 00
F4 0A 00 01 00 01 00 00 66 FF 00 00 00 00 00 00

welash, 2017-06-07:
Based on your comparison of the dumps, it looks to me like what you tried should have worked. (that is supplying the EX(start) as the byte before the HI part of the address.  So it seems like if you changed the F40A and F40B commands to add that byte, things might "just work".  

The only thing that looks a bit different is that they seem to have an extra FF in the buffer after the first byte.  If you try reading 16 bits at 0xfffe and 0x10000 do they also append an FF

janezNov, 2017-06-08:
I cleaned comparison document a bit, deleted wait states 01 00 00 00, ? led toggle ?
Next I will try to modify and compile stm8flash code to modify one byte in 0x10000 address, hardcoded

[20170608_1byte_0x66_write_0xffff_0x10000_comparison.xlsx](https://github.com/vdudouyt/stm8flash/files/1060320/20170608_1byte_0x66_write_0xffff_0x10000_comparison.xlsx)

janezNov, 2017-06-08:
Good news.
I managed to write 1byte on address 0x10000
[image]

janezNov, 2017-06-09:
Writing to address over 64k works in latest commit:
https://github.com/janezNov/stm8flash/commit/83040e2e5b88fa036ee33bf0288f0de3b8ddd308
Verification not working for now

spth, 2017-08-16:
Closing this issue. The fix has been merged, and I verified it works (see updated README in pull request #74)

Philipp

closed 2017-08-16 by spth