vdudouyt/stm8flash #37: strange pause STLinkv2 and OS X

kensenjiha, 2016-01-12:
in OS X 10.11.1, compiled stm8flash with libusb1.0.20 (homebrew)

```
# make
cc `pkg-config --cflags libusb-1.0` -g -O0 --std=gnu99 --pedantic -I`xcrun --show-sdk-path`/usr/include/ -I`xcrun --show-sdk-path`/usr/include/sys -I`xcrun --show-sdk-path`/usr/include/machine   -c -o stlink.o stlink.c
cc `pkg-config --cflags libusb-1.0` -g -O0 --std=gnu99 --pedantic -I`xcrun --show-sdk-path`/usr/include/ -I`xcrun --show-sdk-path`/usr/include/sys -I`xcrun --show-sdk-path`/usr/include/machine   -c -o stlinkv2.o stlinkv2.c
cc `pkg-config --cflags libusb-1.0` -g -O0 --std=gnu99 --pedantic -I`xcrun --show-sdk-path`/usr/include/ -I`xcrun --show-sdk-path`/usr/include/sys -I`xcrun --show-sdk-path`/usr/include/machine   -c -o main.o main.c
main.c:239:10: warning: enumeration value 'UNKNOWN' not handled in switch [-Wswitch]
                switch(memtype) {
                       ^
1 warning generated.
cc `pkg-config --cflags libusb-1.0` -g -O0 --std=gnu99 --pedantic -I`xcrun --show-sdk-path`/usr/include/ -I`xcrun --show-sdk-path`/usr/include/sys -I`xcrun --show-sdk-path`/usr/include/machine   -c -o byte_utils.o byte_utils.c
cc `pkg-config --cflags libusb-1.0` -g -O0 --std=gnu99 --pedantic -I`xcrun --show-sdk-path`/usr/include/ -I`xcrun --show-sdk-path`/usr/include/sys -I`xcrun --show-sdk-path`/usr/include/machine   -c -o ihex.o ihex.c
cc `pkg-config --cflags libusb-1.0` -g -O0 --std=gnu99 --pedantic -I`xcrun --show-sdk-path`/usr/include/ -I`xcrun --show-sdk-path`/usr/include/sys -I`xcrun --show-sdk-path`/usr/include/machine   -c -o stm8.o stm8.c
cc stlink.o stlinkv2.o main.o byte_utils.o ihex.o stm8.o `pkg-config --libs libusb-1.0` -o stm8flash
# sudo make install 
cp stm8flash /opt/bin/
```

has the following issues:
* 1. there's a 3-second pause before read/write operation, like this:

```
# stm8flash -c stlinkv2 -p stm8s003f3 -s eeprom -r eep.bin 
Determine EEPROM area
[pause here for ~3 seconds]
Reading 128 bytes at 0x4000... OK
Bytes received: 128
```

* 2. after one read or write, (either successful or failure), next read/write will stuck like this:

```
stm8flash -c stlinkv2 -p stm8s003f3 -s eeprom -r eep.bin 
Determine EEPROM area
[stuck here forever, until STLinkv2 is unplugged]
libusb: warning [darwin_transfer_status] transfer error: device not responding (value = 0xe00002ed)
IO error: expected 2 bytes but 0 bytes transferred
```

unplug STLinkv2 and plug again, flash goes well (still has mentioned ~3-second pause)

Additional info (might be relevant):
[st-flash](https://github.com/texane/stlink) suffers from pause issue also, between 3-8 seconds, but no issue 2. on OS X 10.11 and 10.10

kensenjiha, 2016-01-14:
update:
Updated OS X to 10.11.2 yesterday, then the pause has magically gone. Guess Apple fixed something. . .
Still stuck there after one read/write operation.

nhantt92, 2017-08-11:
Hi, I have error when make, help me

nhantts-MacBook-Pro:stm8flash nhantt$ make
make: pkg-config: Command not found
make: pkg-config: Command not found
cc -g -O0 --std=gnu99 --pedantic   stlink.o stlinkv2.o main.o byte_utils.o ihex.o srec.o stm8.o  -o stm8flash
Undefined symbols for architecture x86_64:
  "_libusb_bulk_transfer", referenced from:
      _stlink_send_message in stlink.o
      _stlink_read in stlink.o
      _stlink_send_cbw in stlink.o
      _stlink_read_csw in stlink.o
      _stlink_swim_write_block in stlink.o
      _msg_transfer in stlinkv2.o
  "_libusb_claim_interface", referenced from:
      _usb_init in main.o
  "_libusb_detach_kernel_driver", referenced from:
      _usb_init in main.o
  "_libusb_exit", referenced from:
      _stlink_close in stlink.o
  "_libusb_free_device_list", referenced from:
      _usb_init in main.o
  "_libusb_get_device_list", referenced from:
      _usb_init in main.o
  "_libusb_init", referenced from:
      _usb_init in main.o
  "_libusb_kernel_driver_active", referenced from:
      _usb_init in main.o
  "_libusb_open_device_with_vid_pid", referenced from:
      _usb_init in main.o
  "_libusb_set_debug", referenced from:
      _usb_init in main.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [stm8flash] Error 1
nhantts-MacBook-Pro:stm8flash nhantt$

phult-ad, 2017-08-15:
Hi, nhantt92.
You should install pkg-config and libusb via brew:
brew install pkg-config
brew install libusb

I hope this can help you!

spth, 2021-09-19:
Looks like this is out of date. In the unlikely case that it is not, the in-progress tlinkv2 driver rewrite will surely make it out of date very soon.

closed 2021-09-19 by spth