vdudouyt/stm8flash #46: STM8S903K3 not working

kubilho, 2016-06-13:
Hello to All.

STM8S903K3 with stlinkv2 and stm8flash results:

Determine FLASH area
Writing Intel hex file 2550 bytes at 0x8000... Tries exceeded

With stlink flashing suspends. It is done after few Bytes. I found out this by STVP.

Chip is not write protected and is unlocked - checked by STVP.

Any reasons?

spth, 2016-06-24:
I think the flash_block_size for the stm8s903 is wrong in stm8.c. Please try if it works when you change it to 64.

Philipp

kubilho, 2016-06-24:
Thanks Philipp.
It is working now :)
On monday I will check all of stm8flash features with stm8s903.

spth, 2016-06-24:
I created pull request #47 for the fix.

Philipp

kubilho, 2016-06-27:
Philipp,

I have tested stm8s903?3 with stlinkv2
OPT: OK.
Flash: OK.
Eeprom: OK.
Unfortunately I have no time to organise stlinkv1.

closed 2016-06-27 by kubilho