vdudouyt/stm8flash #57: iSerial of programmer gets corrupted after one attempt to write

adithyaselv, 2017-03-08:
Hi,
I have been trying to program multiple boards connected to single pc. I tried to use an additional argument iSerialNumber of the device to distinguish between multiple programmers. But when I run your flash code , the iserial number of the programmer seem to get corrupted after one attempt and i am unable to use that for the next time(I have plug out the device and plug in again to get correct serial number). If I just try to open and close the usb port using libusb , the number doesn't get corrupted. Does your code access device config parameters somewhere ? or changes it somewhere while writing into STM flash ? Kindly help

lkord, 2018-04-24:
@adithyaselv Can You tell how to use iSerial? Maybe some code example?

adithyaselv, 2018-07-01:
wow great thanks

closed 2018-07-01 by adithyaselv