vdudouyt/stm8flash #61: MacOSX Yosemite - error: use of undeclared identifier 'stderr'

e-onux, 2017-04-08:
make
cc -g -O0 --std=gnu99 --pedantic -I/usr/include/ -I/usr/include/sys -I/usr/include/machine -I/usr/local/Cellar/libusb/1.0.21/include/libusb-1.0    -c -o stlink.o stlink.c
stlink.c:250:3: warning: implicit declaration of function 'usleep' is invalid in
      C99 [-Wimplicit-function-declaration]
                usleep(10000);
                ^
stlink.c:345:4: warning: implicit declaration of function 'fprintf' is invalid
      in C99 [-Wimplicit-function-declaration]
                        fprintf(stderr, "Unknown status: %x\n", status);
                        ^
stlink.c:345:4: warning: declaration of built-in function 'fprintf' requires
      inclusion of the header <stdio.h> [-Wbuiltin-requires-header]
stlink.c:345:12: error: use of undeclared identifier 'stderr'
                        fprintf(stderr, "Unknown status: %x\n", status);
                                ^
stlink.c:545:12: error: use of undeclared identifier 'stderr'
                        fprintf(stderr, "Write error\n");
                                ^
3 warnings and 2 errors generated.
make: *** [stlink.o] Error 1

e-onux, 2017-04-08:
xcode commandline tools already installed. But it throws error.

KenjutsuGH, 2017-06-09:
I am having the same error on **macOS Sierra 10.12.5.**

**xcode-select --install** gives _xcode-select: error: command line tools are already installed, use "Software Update" to install updates_

**gcc -v** gives:
```
Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 8.1.0 (clang-802.0.42)
Target: x86_64-apple-darwin16.6.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
```

spth, 2017-07-03:
I do see an #include <stdio.h> in the current stlink.c, so use of frpintf snd stderr shouldbe fine. Is this problem still there in the current version?

Philipp

KenjutsuGH, 2017-07-05:
Yes, it is still a problem, even with `#include <stdio.h>`. I was able to resolve the problem and get it compiling, and working:

- Removed macOS SDK include directory dependency in `Makefile`
- Added `#ifndef `to remove inclusion of `endian.h` and `malloc.h` in `stlink.c` and `pgm.h`

It still compiles under Linux Mint 18.1, but I was not able to test it under Windows

spth, 2018-01-19:
You fixes have been merged in July.

Philipp

closed 2018-01-19 by spth