vdudouyt/stm8flash #77: 'make install' gives 'no permission' error on macOS

fdx1601, 2017-08-31:
on macOS "Sierra" 10.12.6 (16G29) the "make install" gives an error stating it has no permission to copy stm8flash:

```
localhost:stm8flash ts$ make install
mkdir -p /usr/bin/
cp stm8flash /usr/bin/
cp: /usr/bin/stm8flash: Operation not permitted
make: *** [install] Error 1
```

Adding a 'Darwin' condition to the install target in the Makefile resolves this issue:

```
$ git diff Makefile 
diff --git a/Makefile b/Makefile
index bc1a9c2..b74fc15 100644
--- a/Makefile
+++ b/Makefile
@@ -65,7 +65,11 @@ install:
 ifeq ($(PLATFORM),FreeBSD)
        mkdir -p $(DESTDIR)/usr/local/bin/
        cp $(BIN)$(BIN_SUFFIX) $(DESTDIR)/usr/local/bin/
+else ifeq ($(PLATFORM),Darwin)
+       mkdir -p $(DESTDIR)/usr/local/bin/
+       cp $(BIN)$(BIN_SUFFIX) $(DESTDIR)/usr/local/bin/
 else
        mkdir -p $(DESTDIR)/usr/bin/
        cp $(BIN)$(BIN_SUFFIX) $(DESTDIR)/usr/bin/
 endif
```

Regards.

vbauer, 2017-12-14:
+1 I have the same issue

spth, 2018-01-19:
Isn't /usr/local/bin the conventional place on other OSes, such as GNU/Linux and OpenBSD anyway? Shouldn't we just make installation into /usr/local/bin/ the default for all OSes?

Philipp

spth, 2018-01-22:
I now made /usr/local/bin/ the default install location for all platforms.

Philipp

closed 2018-01-22 by spth

vbauer, 2018-01-23:
Now it works fine, thank you!