vdudouyt/stm8flash #78: Getting "Tries exceeded" with ST-LinkV2 clone

fdx1601, 2017-08-31:
Hi folks!

Since a couple of days I wasn't able to read or write to any of my ESP14 modules with "stm8s003f3 inside".

Occasionally it seemed to work very few times ... but then stalled again getting the "Tries exceeded".

Suggestions given here in other posts didn't work either. 

Totally stuck I had a look into the sources ... especially the ``try.h`` file. Accordingly tu the commit history the delay value has been changed recently from 3000us to 10000us in order to get higher rate of success using the ST-LinkV2 clones.

Since mine was not working at all I decided to go back to the original 3000us. After a quick build I've to say that it started to work. From the very first try ... alter having failed with the original 10000us before.

I don't know if this may be circuit related, platform or usblib related ... but the important thing for now is that it works and I can proceed exploring the STM8 devices.

In order to be able to adjust the delay values I added an ``-d`` option under ``Troubleshooting`` section which allows to choose between ``default`` (3ms), ``slow`` (10ms) or even a numeric value in ``ms``.

The diff to the original source is attached below. This patch also corrects the ``no permission`` issue from my previous post when running ``[sudo] make install``.

[stm8flash.diff.txt](https://github.com/vdudouyt/stm8flash/files/1267942/stm8flash.diff.txt)

Regards.

liangx8, 2017-09-21:
I have this issue too, when I got this. I switch to windows and using stv programmer. connetion still not work. I was prompted connection cable problem. But after I plug and unplug few times, it work again both in windows and linux. I guess it's issue of stlinkv2 clone's firmware.

liangx8, 2017-09-21:
After I do testing operation. I found some pattern. the access operation(read/write flash/option bytes) only works under target board is halt. I do follow sequence to halt target
Do any access while keep nrest pin at target board low, and get a Tries exceeded error. then release nrest pin.

fdx1601, 2017-11-05:
Hi today I found the time to get back to this. Magically everything works now. Perhaps things have been fixed by one of the macOS updates in the meantime. 

The only time it didn't work was when the GND jumper cable slipped off the contact ;-)

Just a side note: currently I am playing around with some of these ESP14 boards with integrated STM8S003F3. Now that I am able to flash the STM8 we'll see how to make it speak to the ESP.

Regards.

mishazawa, 2020-10-16:
This issue was gone when I change GND pin.

embhobbb, 2020-10-25:
I am also getting the same issue. i am using the stm8s103 developement board and stlink v2 clone.
i have connected usb power connection for power and stlink to the board.
i had a working setup before but after windows format i am getting this issue.  The following are the logs i am getting.

```
stm8flash -c stlinkv2 -p stm8s103f3 -w blink.ihx
Determine FLASH area

Due to its file extension (or lack thereof), "blink.ihx" is considered as INTEL HEX format!
455 bytes at 0x8000... Tries exceeded

```

mmilch, 2020-10-25:
I remember I had similar issue with Intel HEX format. There are two solutions:

- you can flash Intel Hex file by ST Visual Programmer (STVP) into program memory
[](https://www.st.com/en/development-tools/stvp-stm32.html)
or
- you can use HEX2BIN converter from here and flash by stm8flash converted file:
[](https://www.keil.com/download/docs/7.asp)

STLINKV2 is surely working with STVP if you choose STM8S105x4 driver built in it (checked!)

I suggest also making direct connection from STLINKV2 into STM8 chip and also supply it from STLINKV2
(without standard board power supply) according to scheme:
- stlink SWIM pin = STM8 chip SWIM
- stlink RST pin = STM8 chip NRST
- stlink GND pin = STM8 chip VSS
- stlink +5V pin = STM8 chip VDD

embhobbb, 2020-10-25:
@mmilch i tried allt he ways but still the result is not expected.
Again i tried it on linux but getting same issue. 
i tried 2 stlink and 3 stm8s board. All are giving same problem..
STLInk looks working because i checked with stm32f103 and it was detected in stm32 flasher but getting issue with stm8s103..

mmilch, 2020-10-25:
https://community.st.com/s/question/0D50X0000AqB8JuSQK/stm8s103f3-getting-tries-exceeded-with-stlink-ubuntu

Check with multimeter are all connections between chip pins and development board (same pins) "OK".
It might be some board or chip internal issue.

Try also to clear ROP byte and then check - is behavior the same:

$ echo "00" | xxd -r -p >ROP_CLEAR.bin
$ stm8flash -c stlinkv2 -p stm8s103f3 -s opt -w ROP_CLEAR.bin

Nable80, 2020-10-25:
Some MCUs arrive locked and programmed with factory test firmware. I don't remember which error I got when trying to flash them but they were fixed by erasing (``-u``): ``stm8flash -c stlinkv2 -p 'stm8s103?3' -u``

I've successfully uploaded firmware after that with the following command: ``stm8flash -c stlinkv2 -p 'stm8s103?3' -s flash -w main.ihx``

Probably you should also check that you supply power to the board (ST-Link dongles that I saw only measure Vcc line, they don't try to feed the target) and capacitors at Vcc, Vcap and nRST have proper values (nRST's one shouldn't be very large, others should be large enough).

embhobbb, 2020-10-27:
i found one more st-link clone and stm8s working with that..i removed the case and checked the hardware. On high level it looks very similar. So i think the issue is with the stlink firmware version. 
I think older firmware version are working fine with stm8s..
Now i want to know which firmware version working fine with stm8s...
i will be uopdating result here once i get it...

thinhx2, 2020-11-19:
Hello, yesterday i got stlink v2 clone and try to flash a stm8f103 breakout board i bought few week ago. I have same problem with all, chip protect,try exceeded, and a common problem of stm8 china board - gnd pin at debug header is not connected to mcu, i changed to another gnd port. I was think i solded fail and make debug header is damage. But today im wrong, i re-check pinout of breakout board and i see D1 can be use as swim, i try use that port but when i connect header of stlink to jumper, reset and swim connect is pushed inwards. So, i opened it and resolder swim and reset connect, seem like a mistake when employee sold it, the metal part used to connect the plug has been tin soldered but it has not been solder to the board, the distance is about 0.05mm.  I soldered more tin so it can connect to the circuit board.  And after that, everything worked fine. Sorry, my english is not good, hope all friend cam understand it

[image]

prosper00, 2020-12-05:
I also have some stm8s103f3 boards with un connected ground pins on the programming header. My solution:
[image]

thinhx2, 2020-12-05:
> I also have some stm8s103f3 boards with un connected ground pins on the programming header. My solution:
> [image]

Did you scrape the green paint on the copper layer? welding directly from gnd to amss is ok?  I'm scared when I scrape the blue paint so hard that the circuit breaks

prosper00, 2020-12-05:
yeah, I just scraped off a bit of the solder mask till I could see a tiny spot of copper, then tinned it and soldered a piece of thin magnet wire to that pad. Then I secured things with a dab of krazy glue

jay-ramani, 2022-03-27:
> i found one more st-link clone and stm8s working with that..i removed the case and checked the hardware. On high level it looks very similar. So i think the issue is with the stlink firmware [...]

Did you manage to find out which f/w works? I've run into the same issue and none of the solutions posted here work for me :-(