vdudouyt/stm8flash #81: Verify option for ihx is broken

rumpeltux, 2017-10-25:
I’m not sure exactly how though. But this seems to already have been spotted in #14 
Verify for binary files works.

welash, 2017-11-27:
Issue #14 was reported before intel hex format was supported.  I will try to take a look at this later, but just to make sure, the file that you are verifying from is an intel hex file, and it has an extension of either ".hex" or ".ihx", correct.  Any other extension will be treated as a binary file.  We probably should use a command line option or other method for determining the file type, but we don't/

rumpeltux, 2017-12-02:
Yes, I had the correct extension, verifing .bin or any other works fine.

welash, 2017-12-03:
Ok,  I think I see the problem. the call to ihex_read is passing in buf instead of buf2.  The fix should be:
```
diff --git a/main.c b/main.c
index df3b0db..2728cee 100644
--- a/main.c
+++ b/main.c
@@ -371,7 +371,7 @@ int main(int argc, char **argv) {
                int bytes_to_verify;
                /* reading bytes to RAM */
                if(is_ext(filename, ".ihx") || is_ext(filename, ".hex")) {
-                       bytes_to_verify = ihex_read(f, buf, start, start + bytes_count);
+                       bytes_to_verify = ihex_read(f, buf2, start, start + bytes_count);
                } else {
                        fseek(f, 0L, SEEK_END);
                        bytes_to_verify = ftell(f);
```
But I haven't had a chance to test it.

rumpeltux, 2017-12-05:
I tested it. Seems to work :) Thanks.

welash, 2017-12-05:
Great,  I'll submit a pull request when I get a chance.

spth, 2018-01-19:
Clsoing the issue, since the fix has been merged and tested.

Philipp

closed 2018-01-19 by spth