vdudouyt/stm8flash #90: ST-Link/v3 support

spth, 2018-01-11:
Apparently, ST is about to release version 3 of the ST-Link:

https://community.st.com/thread/46479-stlinkv3

Once sufficient documentation becomes available, stm8flash should add support.

Philipp

spth, 2018-09-19:
Here is a document comparing the various ST-Link versions:

TN1235: Overview of ST-LINK derivatives (https://www.st.com/resource/en/technical_note/dm00290229.pdf)

Apparently, relevant to us are ST-LINK/V1, ST-LINK/V2 and STLINK-V3.

There also are ST-LINK/V2-1, ST-LINK/V2-A and ST-LINK/V2-B, but those are for the STM32 only not for the STM8 (they do not have a SWIM interface).

Philipp

spth, 2018-09-27:
Confusingly, while the document states that the ST-LINK/V2-1 does not support SWIM (which would mean it does not support the STM8), ST has recently released the STM8 Nucleo boards, and the documentation states that they have an integrated ST-LINK/V2-1.

Philipp

martonmiklos, 2018-09-27:
It seems that mouser has it on stock now:
https://hu.mouser.com/ProductDetail/STMicroelectronics/STLINK-V3SET

spth, 2018-09-27:
Thanks. The picture at Mouser, however apparently shows an old ST-LINK/V1. And according to documentation, the ST-LINK/V3 requires an adapter board MB1440 when used with the STM8. I hope it is included in the STLINK-V3SET offered by Mouser.

Philipp

martonmiklos, 2018-09-27:
Ah this would not be the first time when mouser product image would be wrong :D.

martonmiklos, 2018-10-02:
@spth 
Good news: MB1440 is included in the kit. 
I will post details about the device soon.

martonmiklos, 2018-10-03:
I have put the teardown images here including the USB descriptors:
http://users.atw.hu/balubati/blog/index.php?entry=entry181002-114748

martonmiklos, 2018-10-25:
The support the device is landing for the openOCD:
http://openocd.zylin.com/#/c/4717/

I have contacted with the author regarding to the docs and he said:

> I have received some documentation from ST, but under "non disclosure
> agreement" and specifically for extending the support of OpenOCD.
> Sorry, I cannot share the documentation with you, nor I can use it for
> another project.
> But my contribution to OpenOCD is under GPLv2, so you are free to
> re-use my code in another project under the same license as stm8flash.
> 
> Till now I had no possibility to verify that my port to OpenOCD works
> with STM8. I will probably try next week. Maybe we can update after
> that. And anyway I will push the changes/fixes in OpenOCD.
> Anyway, from the documentation I have, I do not see any change between
> ST-Link/V2 and STLINK-V3 concerning SWIM.
> I expect it should work straight-forward.
> If you really want to extend stm8flash, I think you can simply have a
> look to the last patch in my series
> http://openocd.zylin.com/4717
> you need:
> - the new USB PID,
> - the new command for version check,
> - and you can skip the new commands for speed, specific for JTAG and
> SWD (not used for SWIM)
>