vdudouyt/stm8flash #91: No success flashing STM8L001J3

TG9541, 2018-01-14:
I tried flashing the STM8L001J3 with binary contained in vdudouyt/stm8flash
 3e034d3 (master, Dec 13, 2017). Flashing the device doesn't work for unknown reasons. I tried using the "-d slow" option to no avail. Setting the option bits, however, works. Note that the device doesn't have a NRST pin.

Note:
I had a look at the code (which doesn't compile for me), and found that stm8flash doesn't cluster technical information about STM8 variants into family and density. It simply uses the marketing name, and leaves QA to the user (the list in README.md).

Also note that I published information, derived from the STM8CubeMX  `db/mcu` folder, on hypothetical common silicon properties in the [Wiki of my fork of this repository](https://github.com/TG9541/stm8flash/wiki).

TG9541, 2018-01-14:
I did some research: 

* [AN2659 STM8 in-application programming (IAP)](http://www.st.com/content/ccc/resource/technical/document/application_note/0d/59/82/a6/87/d5/4f/3b/CD00176595.pdf/files/CD00176595.pdf/jcr:content/translations/en.CD00176595.pdf) delegates information about flash programming to the STM8 Standard Peripheral Library
* STM8FLASH uses block programming for Flash
* [STSW-STM8012 STM8L10x standard peripheral library](https://my.st.com/content/my_st_com/en/products/embedded-software/mcus-embedded-software/stm8-embedded-software/stsw-stm8012.license%3d1515861696920.html#design-scroll), unlike the other STM8L and STM8S libraries, doesn't provide a block programming function
* the FLASH_xx registers are defined in the STM8L10xx datasheet, and I couldn't find any information about differences between the STM8L10xx sub-family and the other STM8L and STM8S devices.

I added a [log facility](https://github.com/rxi/log.c) to my local copy of STM8FLASH, and made it write an interaction trace. It fails whenever I use `stlink2_wait_until_transfer_completes()`. In block mode it didn't write any data. However, a test with writing Flash data in byte mode wrote the first 4 bytes to my STM8L001J3 chip.

icraftcrafts, 2018-06-21:
hi.i just tried the raw chip. it seems to flash when power and programmer power are present. its possibly to do with the delay that needs to be there with the circuit power before the programmer uses swim. this definitely works on stm8flash. ive created a win vm with cosmic ​and got an error in there that led me to try this.