vdudouyt/stm8flash #94: Fail to build under Windows with mingw64

eeyrw, 2018-01-23:
`PS C:\Users\admin\Desktop\stm8flash> make -j5
GCC -g -O0 --std=gnu99 --pedantic     -c -o stlink.o stlink.c
GCC -g -O0 --std=gnu99 --pedantic     -c -o stlinkv2.o stlinkv2.c
GCC -g -O0 --std=gnu99 --pedantic     -c -o espstlink.o espstlink.c
GCC -g -O0 --std=gnu99 --pedantic     -c -o main.o main.c
espstlink.c:27:10: fatal error: termios.h: No such file or directory
 #include <termios.h>
          ^~~~~~~~~~~
GCC -g -O0 --std=gnu99 --pedantic     -c -o byte_utils.o byte_utils.cco
mpilation terminated.
make.exe": *** [espstlink.o] Error 1
make.exe": *** Waiting for unfinished jobs....
PS C:\Users\admin\Desktop\stm8flash>`

It seems that the code require the header termios.h and is not presneted by mingw64. And what's is espstink? A wifi stlink made by ESP8266??
I use normal usb stlink for STM8s. So can I remove this part code directly?

helmo2004, 2018-01-27:
When a project uses termios.h it seems that the only way to build it under windows is cygwin. I would suggest to use a wrapper for serial port to stay platform-independent. My solution a the moment was to remove espstlink locally, because I don't need it.

Btw: I like the idea of using an esp8266 for flashing STM8-devices. So it would be great if it was available under windows.

Details about espstlink can be found here: [https://github.com/rumpeltux/esp-stlink](url)

Greetings Oliver

eeyrw, 2018-01-28:
I have truncated the espstlink part and got it work and feel surprise that esp8266 stm8 flasher actually exists : https://github.com/lujji/esp-stm8-flasher.

tenbaht, 2018-11-22:
Same problem here. For the [Sduino](https://github.com/tenbaht/sduino) project I need to compile binaries for different systems, including Windows. My solution is to cut out the espstlink code, but using unmodified code would be a far better solution.

I suggest using a simple intermediate cross-platform layer like [libserialport](https://sigrok.org/wiki/Libserialport) or to borrow some code from the [stm8gal](https://github.com/gicking/stm8gal) project.