vdudouyt/stm8flash #96: Failed to write STM8L151K4 (stm8l151?4)

czepiec, 2018-02-10:
Can't write to stm8l151k4 due to block size. In family "stm8l151?4" is block size 128 bytes instead of 64 bytes, please check PM0054 page 26. It is in file stm8.c line 474 (e.g. 72be7ba).

spth, 2018-02-12:
Interesting. Indeed, the programming manual PM0054 states the block size as 128 on page 26.

stm8flash uses a block size of 64, which is the value from the the device datasheet, section 9.3.5, "Memory characteristics".

Have you verified that your device work when the block size is changed to 128 in stm8.c?

Philipp

czepiec, 2018-02-12:
Yes, I tested it. When I read from stm8l151k4 it was not problem, but when I wanted write (-w) to MCU it did not work, program froze. This was output with #debug:
>Determine FLASH area
Due to its file extension (or lack thereof), "hex.ihx" is considered as INTEL HEX format!
16384 bytes at 0x8000...
stlink_swim_write_byte
stlink_swim_write_byte
stlink_swim_write_byte
stlink_swim_write_byte
stlink_swim_write_byte
stlink_swim_write_byte
stlink_swim_write_byte
stlink_swim_write_byte
Writing block 8000 with size 64
stlink_swim_write_byte
Writing block 8040 with size 64
stlink_swim_write_byte

When I changed block size to 128 writes works perfectly.
I think UM0054 is more relevant and in datasheet section 9.3.5, table 36., row 2. and 3 is mistake.

I work a lot with STM devices, later I can check other MCU.

spth, 2018-02-12:
Thanks. I'll change it. Which programmer did you use to read/write the flash?

Philipp

czepiec, 2018-02-12:
Currently I use stlink (v1) on STM8L discovery board (only SWIM interface).

By the way, thanks for this software. I had in past dedicated Win PC for stlink but now I can do everything on my Linux workstation.

spth, 2018-02-14:
Thanks. I made the change. While I contributed to stm8flash, Valentin did most of the work.

Philipp

closed 2018-02-14 by spth