vdudouyt/stm8flash #99: make show error

Rombersoft, 2018-03-23:
```
cc -g -O0 --std=gnu99 --pedantic `pkg-config --cflags libusb-1.0`    -c -o stlink.o stlink.c
Package libusb-1.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `libusb-1.0.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libusb-1.0' found
In file included from stlink.c:17:0:
pgm.h:7:21: fatal error: libusb.h: Немає такого файла або каталогу
compilation terminated.
<вбудований>: спроба виконання рецепта для мети «stlink.o» зазнала невдачі
make: *** [stlink.o] Помилка 1

```

vdudouyt, 2018-03-25:
It looks that you don't have libusb-1.0 development headers installed.

microwiz, 2018-05-05:
In Ubuntu 16.04.4 LTS, I was able to get past this error by installing libusb. I used "sudo apt-get install libusb-1.0" and all was taken care by apt.

This issue can be closed.

closed 2018-06-26 by spth

vksingh18881, 2019-05-26:
In Ubuntu 18.04.2 LTS I have same problem as mentioned by Rombersoft. I have installed installed libusb-1.0-0-dev but error remains the same. libusb.h is in /usr/include/libusb-1.0/
Please kindly help.

Picatout, 2020-02-05:
@vksingh18881,  
I had the same problem in ubuntu 18.04.3 LTS I installed pkg-config and issue was solved.
sudo apt install pkg-config