vdudouyt/stm8flash #100: not compiled in ubuntu 18.04

GavukaAlexandr, 2018-05-13:
sudo apt-get install pkg-config
sudo apt-get install libusb-1.0-0-dev
You need to change the path to the libusb in the `stm8flash/pgm.h` file on `#include </usr/include/libusb-1.0/libusb.h>`

spth, 2018-06-26:
I guess we'd need a configure script or a full autotools setup to handle varying libusb locations portably.

Philipp

mrx23dot, 2021-08-23:
Same problem on debian too, it's located at /usr/include/libusb-1.0/libusb.h

stefaandesmet2003, 2021-09-27:
same file location on ubuntu 20.04, but stm8flash compiles OK.

agentgoblin, 2021-11-16:
Debian 11 and Ubuntu 20.04 reporting. Package libusb-1.0-0-dev places the libusb.h in /usr/include/libusb-1.0/libusb.h in both distros. stm8flash compiled successfully after installing pkg-config.

```
$ make
cc -g -O0 -DDEBUG=0 --std=gnu99 --pedantic -Wall `pkg-config --cflags libusb-1.0`    -c -o stlink.o stlink.c
cc -g -O0 -DDEBUG=0 --std=gnu99 --pedantic -Wall `pkg-config --cflags libusb-1.0`    -c -o stlinkv2.o stlinkv2.c
cc -g -O0 -DDEBUG=0 --std=gnu99 --pedantic -Wall `pkg-config --cflags libusb-1.0`    -c -o espstlink.o espstlink.c
cc -g -O0 -DDEBUG=0 --std=gnu99 --pedantic -Wall `pkg-config --cflags libusb-1.0`    -c -o main.o main.c
cc -g -O0 -DDEBUG=0 --std=gnu99 --pedantic -Wall `pkg-config --cflags libusb-1.0`    -c -o byte_utils.o byte_utils.c
cc -g -O0 -DDEBUG=0 --std=gnu99 --pedantic -Wall `pkg-config --cflags libusb-1.0`    -c -o ihex.o ihex.c
cc -g -O0 -DDEBUG=0 --std=gnu99 --pedantic -Wall `pkg-config --cflags libusb-1.0`    -c -o srec.o srec.c
cc -g -O0 -DDEBUG=0 --std=gnu99 --pedantic -Wall `pkg-config --cflags libusb-1.0`    -c -o stm8.o stm8.c
cc -g -O0 -DDEBUG=0 --std=gnu99 --pedantic -Wall `pkg-config --cflags libusb-1.0`    -c -o libespstlink.o libespstlink.c
cc -g -O0 -DDEBUG=0 --std=gnu99 --pedantic -Wall `pkg-config --cflags libusb-1.0`  stlink.o stlinkv2.o espstlink.o main.o byte_utils.o ihex.o srec.o stm8.o libespstlink.o `pkg-config --libs libusb-1.0` -o stm8flash
```

```
$ ./stm8flash -V
20170616-1.1
```

agentgoblin, 2021-11-16:
> I guess we'd need a configure script or a full autotools setup to handle varying libusb locations portably.

pkg-config does it.

spth, 2021-11-17:
For me (Debian GNJ/Linux testing) it always worked fine via the pkg-config in the Makefile.