vdudouyt/stm8flash #101: Support for multiple programmers

PanKaczka, 2018-05-19:
Currently stm8flash is using libusb_open_device_with_vid_pid function that returns handle to the first device that match vID and pID.
I've created a simple patch that adds -S option that allows for specifying the desired programmer.
I have some difficulties creating pull-request so feel free to use my change if you find it useful.

I can also confirm the tool works flawlessly for target stm8l151?6 (STML151G6 in my case)
[multiple.diff.gz](https://github.com/vdudouyt/stm8flash/files/2019289/multiple.diff.gz)

lkord, 2018-05-24:
Can You give an example of use (format of serial)? How do You get serials of STLInks? For the moment I use st-info --probe from texane st-link

PanKaczka, 2018-05-24:
As far as I know st-info uses similar format of the serial number.
Basically I'm using the iSerialNumber string from libusb device descriptor. I then directly print the char values in hexadecimal uppercase format (some characters in iSerialNumber string are not printable and would be hard to copy-paste). 
Example serial of my stlink-v2-clone: 543F7206757757493431123F

lkord, 2018-05-24:
@PanKaczka I think that's the case:
`// print programmer data if no serial specified
+				if(!pgm_serial_specified) {
+					fprintf(stderr, "Programmer %d: %s %s, Serial:%s\n", numOfProgrammers, vendor, device, serial_hex);
+				}
+				else  **//<-- no { here**
+					// otherwise check if it's the correct one
+					if(0==strcmp(serial_hex, pgm_serial)) {
+						pgm->dev_handle = tempHandle;
+						break;
+					}
+				}
+				libusb_close(tempHandle);`

spth, 2018-10-03:
Support for multiple programmers has been merged.

Philipp

closed 2018-10-03 by spth