vdudouyt/stm8flash #102: stm8af5288: Can't flash

AokiAhishatsu, 2018-05-23:
`$ ./stm8flash -c stlinkv2 -p stm8af528? -s flash2 -w ./myprog.hex
Determine RAM area
Due to its file extension (or lack thereof), "./myprog.hex" is considered as INTEL HEX format!
Address 8000 + 4 is out of range at line 1
`
ST Visual Develop it works without problems.

AokiAhishatsu, 2018-05-24:
STM8A52F88

`$ ./stm8flash -c stlinkv2 -p stm8af528? -s flash2 -w ./myprog.hex` (no -s flash2)
Works now.

Weird thing now is, if I read flash it stops the program execution on MCU and is not able to run again until write again (with the abobe command)

GavukaAlexandr, 2018-05-24:
try unlocking the controller:
`echo "00 00 ff 00 ff 00 ff 00 ff 00 ff" | xxd -r -p > factory_defaults.bin
stm8flash -c stlinkv2 -p stm8s103f3 -s opt -w factory_defaults.bin
`

AokiAhishatsu, 2018-05-24:
thx GavukaAlexandr! That works.

Why is this necessary after read/verify?

GavukaAlexandr, 2018-05-24:
You are welcome AokiAhishatsu. 
Because the new stm8 is blocked for reading

spth, 2018-05-26:
flash2 is not  a valid memory type or address. stm8flash should have given an error message; I'll leave this item open until that bug is fixed.

Philipp

spth, 2018-05-26:
Previously, when the parameter to -s was not a known string, instead the longest prefix that is a hex number was used as memory address. for "flash2", this was f.
I now changed it to emit an error when the prefix is not the whole argument.

That's probably still not ideal (there is still potential for error when people assume the location to be in a base other than 16), but better than the old behaviour.

Philipp

closed 2018-05-26 by spth