vdudouyt/stm8flash #116: Use "wb" as fopen flag for Windows compatibility

HiFiPhile, 2018-11-27:
It turns out in Windows, if "w" is used as fopen flag, line break will be added to the file written when fwrite is called. That's why I got a file of 32800+ bytes when reading a 32k device.

> if(!(f = fopen(filename, "wb")))
			spawn_error("Failed to open file");
		switch(fileformat)
		{
		case INTEL_HEX:
			ihex_write(f, buf, start, start+bytes_count);
			break;
		case MOTOROLA_S_RECORD:
			srec_write(f, buf, start, start+bytes_count);
			break;
		default:
			fwrite(buf, 1, bytes_count, f);
		}

spth, 2018-11-27:
If added the "b" when operating on binary files. Please check if this works for you now.

Philipp

HiFiPhile, 2018-11-27:
In my case it solved my issue, I'm using latest Mingw under Windows 10.

closed 2018-11-27 by HiFiPhile