vdudouyt/stm8flash #119: STM8L151K4T6 does not work at all

Xhendos, 2019-06-07:
I am trying to program the STM8L151K4T6. 
I am aware that chips may be locked and read only. That's why I tried to unlock it. 
When I do, I get the following output:
```
$ stm8flash -c stlinkv2 -p stm8l151?4 -u
Determine OPT area
Due to its file extension (or lack thereof), "Workaround" is considered as RAW BINARY format!
Tries exceeded
```

When I try to read the current factory bytes, I get the following output:
```
$ stm8flash -c stlinkv2 -p stm8l151?4 -s opt -r opt.bin
Determine OPT area
Due to its file extension (or lack thereof), "opt.bin" is considered as RAW BINARY format!
Reading 128 bytes at 0x4800... Tries exceeded
```
This means that reading nor writing is not working. 
There's always the `... Tries exceeded` message before `stm8flash` stops.

When I plug in the `stlinkv2` and print my kernel message I get the following output.
```
[  624.326197] usb 1-4: new full-speed USB device number 12 using xhci_hcd
[  624.475434] usb 1-4: New USB device found, idVendor=0483, idProduct=3748
[  624.475441] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  624.475446] usb 1-4: Product: STM32 STLink
[  624.475450] usb 1-4: Manufacturer: STMicroelectronics
[  624.475454] usb 1-4: SerialNumber: QÿhH\xc2\x82WP3"\xc2\x87
```

This is my `/etc/udev/rules.d/49-stlinkv2.rules` file.
```
SUBSYSTEMS=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="3748", \
    MODE:="0666", \
    SYMLINK+="stlinkv2_%n"
```

Does anyone else got issues with the STM8L151K4T6 aswell or is it just me?
I am able to use this stlinkv2 for programming the STM32F103C8T6 so the programmer is working.

Xhendos, 2019-06-08:
After hours of research it turned out that it was a hardware error on my part and not a software error on your part. Adding a 0.1 uF decoupling capacitor solved the issue.

```
$ stm8flash -c stlinkv2 -p stm8l151?4 -u
Determine OPT area
Due to its file extension (or lack thereof), "Workaround" is considered as RAW BINARY format!
Unlocked device. Option bytes reset to default state.
Bytes written: 13
```

closed 2019-06-08 by Xhendos

wfrm, 2020-09-23:
> After hours of research it turned out that it was a hardware error on my part and not a software error on your part. Adding a 0.1 uF decoupling capacitor solved the issue.
>
> ```
> $ stm8flash -c [...]

hi, super, but... capacitor? please where?