vdudouyt/stm8flash #120: global variables in the code cause programming error

Yoda49, 2019-08-01:
When i put some global variables in code - stm8flash show error "Writing Intel hex file Address XXXX is out of range at line XX.

Same error causes with static variables in functions.

TG9541, 2020-01-11:
@Yoda49: are you sure you're looking at the right spot?
@vdudouyt: Valentin, I guess that the issue tracker needs some cleaning. I'd like to suggest you build a support community ;-)

spth, 2021-09-09:
Without further information, there is nothing we can do here.
I assume you write a program in C, compile it with SDCC for the STM8, then try to write the resulting .ihx file to an STM8 device?
If yes, could you provide a minimal code sample, show how you invoke SDCC, which version of SDCC you use, the resulting .ihx file and tell us which device you are targeting using which stm8flash command line?

closed 2021-11-17 by spth