vdudouyt/stm8flash #126: Writes invalid Intel Hex files when reading flash >32KB

basilhussain, 2020-03-03:
When reading flash memory from devices with more than 32KB of flash memory (e.g. STM8S208RB with 128KB), an invalid Intel Hex file is written.

All data records following all but the first 'Extended Linear Address' (ELA) record have incorrect address fields. The problem is that stm8flash seems to be incorrectly including the upper bytes of the address extension when outputting all following addresses.

Some examples:

```
:20FFA000000000000000000000000000000000000000000000000000000000000000000041
:20FFC000000000000000000000000000000000000000000000000000000000000000000021
:20FFE000000000000000000000000000000000000000000000000000000000000000000001
:020000040001F9
:2010000000000000000000000000000000000000000000000000000000000000000000000E0
:2010020000000000000000000000000000000000000000000000000000000000000000000C0
:2010040000000000000000000000000000000000000000000000000000000000000000000A0
```

```
:201FFA000000000000000000000000000000000000000000000000000000000000000000041
:201FFC000000000000000000000000000000000000000000000000000000000000000000021
:201FFE000000000000000000000000000000000000000000000000000000000000000000001
:020000040002F8
:2020000000000000000000000000000000000000000000000000000000000000000000000E0
:2020020000000000000000000000000000000000000000000000000000000000000000000C0
:2020040000000000000000000000000000000000000000000000000000000000000000000A0
```

Note that after the ELA records, there are **5** digits for the address - e.g. "**1**0020".

I believe the problem lies in [line 133 of ihex.c](https://github.com/vdudouyt/stm8flash/blob/3ad9500a04c21f08182439425b19d900c2315217/ihex.c#L133), where `chunk_start` should probably be changed to `chunk_start & 0xffff`. I have not tested this, though.

spth, 2020-05-20:
Could you test your proposed fix and file a pull request?

basilhussain, 2020-05-20:
I tried to test the fix back when I first filed this issue, but unfortunately I didn't have any success attempting to build stm8flash on Windows (with MinGW). Mostly down to issues with the makefile, I think, but I don't know enough about make, etc. to know exactly what was wrong or to fix it. Sorry.

basilhussain, 2020-06-05:
I managed to set up a Linux build environment and make this fix. See pull request #129.

By the way, I think the README file could do with some additions describing how to build, and the prerequisites for doing so. I found you cannot build stm8flash on a default Debian install without first installing `libusb-1.0-0-dev` and `pkg-config` packages.

closed 2020-06-07 by basilhussain