vdudouyt/stm8flash #132: Error while comparing S5 record

konstz43, 2021-01-06:
When i try to flash target MCU from Motorola S-record file, i get the error:
`Error while comparing S5 record (number of data records: 1048812) with the number actually read: 4096`
The file is correct and was previously created by the stm8flash. S5 record is: 
`S5031000EC`
Total of S2 records in my file 4096.

I think the bug is on line 95 of srec.c:
`expected_data_records = chunk_addr;`

`chunk_addr` is 3 bytes (8 digits: `sscanf(line, "S%01x%02x%08x", &chunk_type, &chunk_len, &chunk_addr)` on line 75) but `expected_data_records` must be 2 byte only.
Therefore `expected_data_records` is assigned the wrong value (in my case 0x1000EC instead 0x1000).

schneidersoft, 2021-05-09:
Can you provide a file that causes this error.

msalau, 2022-03-28:
@konstz43 
There is a chance that https://github.com/vdudouyt/stm8flash/pull/118 has fixed the issue for you. The PR has already been merged, so try the latest version.

konstz43, 2023-04-07:
Yes it fixed this issue. Thank you.

closed 2023-04-07 by konstz43