vdudouyt/stm8flash #133: Wrong identification when using more tahn two ST-link v2

szlovak, 2021-01-19:
When using more than two programmers you could get:

-duplicated serial numbers
-invalid serial numbers, too short, garbage data

I use four of them to speed up programming process on raspberry PI. After re-plugging programmers it starts to work, but sometimes serial numbers get duplicated. Once I had duplicated twice, so three of them were the same. Please fix it or tell what to check in code. It can be related to hardware itself and when on boot time enumeration happens, maybe, but for sure stm8flash messes up serial numbers

0ktav1us, 2021-01-20:
I have the same problem after flashingю  I am using two flashers
**Before firmware**
```
` stm8flash -c stlink -p stm8l151?4 -u
Determine OPT area
WARNING: More than one programmer found but no serial number given. Programmer 1 will be used:
Programmer 1: STMicroelectronics STM32 Mass Storage, Serial Number:303030303030303030303031
Programmer 2: STMicroelectronics STM32 STLink, Serial Number:303F68064D53373342401143
Due to its file extension (or lack thereof), "Workaround" is considered as RAW BINARY format!
Unlocked device. Option bytes reset to default state.
Bytes written: 13
# stm8flash -c stlink -p stm8l151?4 -S 303F68064D53373342401143 -u
Determine OPT area
Due to its file extension (or lack thereof), "Workaround" is considered as RAW BINARY format!
Unlocked device. Option bytes reset to default state.
Bytes written: 13` 
```
**After firmware**

```
` # stm8flash -c stlink -p stm8l151?4 -u
Determine OPT area
WARNING: More than one programmer found but no serial number given. Programmer 1 will be used:
Programmer 1: STMicroelectronics STM32 Mass Storage, Serial Number:303030303030303030303031
Programmer 2: STMicroelectronics STM32 Mass Storage, Serial Number:303030303030303030303031
Due to its file extension (or lack thereof), "Workaround" is considered as RAW BINARY format!
Unlocked device. Option bytes reset to default state.
Bytes written: 13` 
```
Please help us.

schneidersoft, 2022-11-20:
There seems to be an issue with how the serial numbers are represented in stm8flash.
When the serial number contains a '\0' character the serial number is not converted to hex correctly.
subsequently the printout and selection of the programmer can be erroneous.
see also #134

schneidersoft, 2023-04-26:
actually there are two problems here. First that the old versions of stlink fw have a bug that makes them output 303030303.... and second that the stm8 had a problem when reading out serial numbers containing zero bytes.
should be fixed now. so we can close this

spth, 2023-04-27:
Yes.

closed 2023-04-27 by spth

szlovak, 2023-05-24:
Problem solved, it was none of these. It was related to poor power supply