vdudouyt/stm8flash #134: Multiple Boards with same Serial Number

gicking, 2021-02-20:
I need to reflash multiple [STM8S Discovery](https://www.st.com/en/evaluation-tools/stm8s-discovery.html) boards within the same circuit. I expected that I can select the respective board via -S option. But both STLink are shown with the same serial number (see below), so -S option has no other effect than avoiding the warning. Specifically, even with -S option (and same serial number), "Programmer 1" is always used.

```
Determine FLASH area
WARNING: More than one programmer found but no serial number given. Programmer 1 will be used:
Programmer 1: STMicroelectronics STM32 Mass Storage, Serial Number:303030303030303030303031
Programmer 2: STMicroelectronics STM32 Mass Storage, Serial Number:303030303030303030303031
Due to its file extension (or lack thereof), "fast.ihx" is considered as INTEL HEX format!
115 bytes at 0x8000... OK
Bytes written: 115

```

I understand from [here](https://github.com/vdudouyt/stm8flash/issues/101) that the "-S" option uses USB vendor and procuct IDs to create a serial number. Unfortunately these are not unique if identical programmers are used. How about to also include the "Device" number from lsusb (see below). Apparently this is unique, even if VID and PID are identical

```
Bus 001 Device 008: ID 0483:3744 STMicroelectronics STLINK Pseudo disk
Bus 001 Device 007: ID 0483:3744 STMicroelectronics STLINK Pseudo disk

```

spth, 2021-02-20:
On the other hand, VID and PID are permant. AFAIK the "Device" number can change if you unplug and replug the device.

gicking, 2021-02-20:
that is correct. But as said I cannot address individual STM8S Discovery boards as they have the same VID and PID. The device number from `lsusb` was just a thought. Maybe there is a more persistent, unique USB identifier...?

spth, 2021-02-20:
I wonder if there is an ST-Link-specific identifier that we could use? Maybe the STM32 in it has a serial number that can be read out?

gicking, 2021-02-21:
That would be great! I searched the web today and [this seems to be a viable option](http://wunderkis.de/stlink-serialno/). However, I didn't find a C way to do this, all references were for OpenOCD configuration.

spth, 2021-02-21:
Since linusb -v can display the serial number, then I think stm8flash should be able to get it through libusb.

schneidersoft, 2023-04-09:
any chance some of my pull requests will get merged upstream?

spth, 2023-04-09:
I'm in France for a few days, and have only an STM8A-Discovery here. When I'm back in Germany, assuming I'll find some time, I'll have a look at the pull requests again (and check that multiboard stuff by with multiple ones attached). I put it on my todo list for end of next week.

spth, 2023-04-20:
I also encountered the problem with multiple "303030303030303030303031" serial numbers. However, I only see that number after having used stm8flash once with that programmer. On the first invokation of stm8flash after plugging in an ST-Link/V2, I see different numbers.

spth, 2023-04-20:
In other places it was reported that the "303030303030303030303031" serial numbers after first use are an ST-Link/v2 firmware bug, and that updating the ST-Link/v2 to the latest firmware version will should fix it. I haven't tested that yet myself.

spth, 2024-12-24:
This has been confirmed to be a firmware bug. We even have a warning in current stm8flash when this suspicious serial number is encountered, recommending to upgrade the firmware. Is that good enough to close this issue?