vdudouyt/stm8flash #139: STM8L052C6 - Unable to write to flash and EEPROM, Tries exceeded

VladimirGaristov, 2021-05-19:
Trying to write data to the flash or EEPROM of STM8L052C6 results in `Tries exceeded` error message. Writing to option bytes and RAM works fine and so does reading from all memories. I'm using an unofficial clone of ST-LINK V2. Flashing the same chip with the same st-link works fine with STVP on Windows. The read protection is disabled and I've tried flashing raw binary and intel hex.

tcfkat, 2021-06-14:
Hello Vladimir,
I had exactly the same problem yesterday! Sadly I have no MS-DOS machine here to cross-check. It took me hours to fix the problem. It was a false FLASH block size declaration in **stm8.c**! Change the following, beginning in line 375:
`    {
        .name = "stm8l052c6",
        .ram_start = 0x0000,
        .ram_size = 2*1024,
        .eeprom_start = 0x1000,
        .eeprom_size = 256,
        .flash_start = 0x8000,
        .flash_size = 32*1024,
        .flash_block_size = 128, // was WRONG!
        .option_bytes_size = 13, // was ZERO!
        .read_out_protection_mode = ROP_STM8L, // was ROP_UNKNOWN
        REGS_STM8L
    },`
recompile and it should work. Furthermore I changed at the beginning the value of CLK_CKDIVR:
`// Note: FLASH_NCR2 not present on stm8l
#define REGS_STM8L { \
        .CLK_CKDIVR = 0x50c0,  \
        .FLASH_PUKR = 0x5052,  \
        .FLASH_DUKR = 0x5053,  \
        .FLASH_IAPSR = 0x5054, \
        .FLASH_CR2 = 0x5051,   \
        .FLASH_NCR2 = 0x0000,   \`
but a little bit unsure here, as I am new to STM8.
Finally I updated the firmware of my ST-LINK V2 clone to version **V2J38S7** using a JAR-file from the ST website. This is not related to the problem here, but better save than sorry ...

andyboeh, 2021-07-01:
Thanks for the fix @tcfkat! I ran into the same problem and it's working properly now!

spth, 2021-07-01:
Have option bytes and read-out-protection also been tested with this change?
@andyboeh : did you also make the change in REGS_STM8L tcfkat mentioned, or did it work with just the change to the flash block size?

andyboeh, 2021-07-01:
I did an unlock and it worked - prior to that, flashing was not possible.

No, I only changed the three lines in stm8.c regarding flash_block_size and so on, nothing more. Brand new ST-Link v2 clone with the default firmware.

I can test on further devices, but this is going to take some time (in the order of weeks I suppose, I have to develop the firmware first).

VladimirGaristov, 2021-08-04:
Thank you @tcfkat, I can reconfirm that this fixes the issue. Sorry for taking so long to respond, I had put my project on hold for a while. May I suggest you fork and make a pull request?

spth, 2021-08-29:
Thanks for the fix and confirmations. I've now updated the repo accordingly.

closed 2021-08-29 by spth