vdudouyt/stm8flash #142: add Windows support

mrx23dot, 2021-08-24:
Initial version started:
https://github.com/mrx23dot/stm8flash/pull/new/windows_inst_prep

termios.h /dev/ttyUSB0 COM port handling needs to be replaced with portable one in libespstlink.c

```
GCC -g -O0 --std=gnu99 --pedantic -Wall     -c -o libespstlink.o libespstlink.c
libespstlink.c:30:10: fatal error: termios.h: No such file or directory
   30 | #include <termios.h>
      |          ^~~~~~~~~~~
compilation terminated.
make: *** [libespstlink.o] Error 1

C:\tmp\stm8flash_orig>libespstlink.c

C:\tmp\stm8flash_orig>make
GCC -g -O0 --std=gnu99 --pedantic -Wall     -c -o libespstlink.o libespstlink.c
libespstlink.c: In function 'espstlink_open':
libespstlink.c:59:18: error: storage size of 'tty' isn't known
   59 |   struct termios tty;
      |                  ^~~
```

maxgerhardt, 2022-10-16:
I'm still running into this problem when compiling for Windows -- I have to forecfully comment out the libespstlink stuff to make it compile.

spth, 2023-04-20:
Maybe, until someone make the esp stuff work on windows, we should pick the NO_ESP macro from https://github.com/tenbaht/stm8flash/tree/disable-esp to make compilation for Windows easier.