vdudouyt/stm8flash #152: SWIM error 0x16 when attempting to flash STM8L050J3

ucukertz, 2022-09-05:
Using stm8flash to upload binary to with STM8L050J3 STlink v2 returns the following log:

> stm8flash -c stlinkv2 -p stm8l050j3 -w build/main.ihx
Determine FLASH area
STLink: v2, JTAG: v29, SWIM: v7, VID: 8304, PID: 4837
SWIM error 0x16

I've tried to search for the meaning of the error but I can't seem to find it.
What does the error mean and what can I do to fix it? Thank you!

stefaandesmet2003, 2022-10-19:
Have you been able to figure this out?
You could try enabling debug prints (make stm8flash with DEBUG=1), you'll get more details of the swim commands, and maybe hints as how to solve your problem.
@spth any documentation for these swim errors?

d3xt3r01, 2022-11-17:
+1 for the swim errors. I get the same with a stm8l001j3 which I think it's read protected but I'm not sure.

chaankya, 2023-02-04:
+1 for error 0x16. I am trying to flash stm8s003f3p6. The flash procedure worked earlier but stopped working. Is it due issue with the stlink V2 clones? The stlink V2 is able to flash a stm32 blue pill.

spth, 2023-11-27:
I have no idea about the SWIM errors. Maybe @vdudouyt can help?

consp, 2024-01-24:
In my case the GND/SWIM pins were reversed vs what was on the device, resulting in error 0x16 when effectively using it with DATA (aka SWIM) attatched to GND. Reversing GND/SWIM pins resulted in proper programming.