vdudouyt/stm8flash #156: Does it support compiling exe output under Windows?

wenhuafeng, 2022-10-26:
Does it support compiling exe output under Windows?

allanwmacdonald, 2022-10-26:
I'm not a developer and I am presuming alot here but I would suggest that the stm8flash is purely targeted to run natively under the linux environment.  For windows, I would recommend the freely available stvp toolset from here:  [https://www.st.com/en/development-tools/stvp-stm8.html](url)

wenhuafeng, 2022-10-26:
I hope it can support compiling under Windows, such as mingw64 compiling, and outputting exe to burn the chip with the command line.

joshuadavila, 2022-10-27:
as @allanwmacdonald comments, for windows environment you can use the freely available stvp, check [](https://www.st.com/en/development-tools/stvp-lib-stm8.html) documentation to see how you can burn your chips directly from commandline.

wenhuafeng, 2022-10-27:
@joshuadavila ，Thanks!!!

allanwmacdonald, 2022-10-27:
There is no need for any special compiling or inclusion of libraries.  The stvp toolset comes with a pre-compiled binary command-line executable called "STVP_CmdLine.exe".  That will get your hex file burned to the chip out of the box.  I was unable to find any specific on-line documentation for it but the help is quite comprehensive:
`"C:\Program Files (x86)\STMicroelectronics\st_toolset\stvp\STVP_CmdLine.exe" -help`

joshuadavila, 2022-10-27:
https://www.st.com/en/development-tools/stvp-lib-stm8.html  contains the source code used to create STVP_CmdLine.exe , of course you can use the pre-compiled binary, but it gives you the libraries to embed stvp functionality into your own applications.

wenhuafeng, 2022-10-27:
STVP_ CmdLine.exe test is available, thank you very much!

spth, 2023-04-20:
It is possible to cross-compile stm8flash under GNU/Linux to a Windows .exe.
You'll also need a libusb DLL: https://github.com/libusb/libusb/releases

Pre-compiled Windows binaries (with a libusb DLL included) are provided by the sduino project. Their current Windows tools release is https://github.com/tenbaht/sduino/releases/download/v0.5/sduino-tools_mingw32-2019.02.05.tar.bz2

closed 2023-04-20 by spth