vdudouyt/stm8flash #163: Eclipse IDE support

rtek1000, 2023-04-09:
Hi, I'm using Eclipse to compile STM8 through a compile.sh script that executes the Makefile.

One thing I would like to be able to see when compiling is memory usage (Flash/RAM).

I'm controlling the size of the code through the information that stm8flash returns when programming.

Would it be possible to adapt some plugin for Eclipse to recognize stm8flash in a similar way to what STM32CubeIDE does to recognize ST-Link?

schneidersoft, 2023-04-09:
I'm not sure I understand what you want stm8flash to do. You want it to automatically detect which model of MCU you have connected?

lalo-uy, 2023-04-09:
Ibelive you have to look at the output of the linker.
May be add some command to the Makefile to list the segments and total memory usage.

spth, 2023-04-09:
When using an IDE, it might make more sense to use GDB/OpenOCD instead of stm8flash. stm8flash can only do basic stuff, such as read and write the flash and other memories. GDB/OpenOCD can do on-target debugging.

rtek1000, 2023-04-09:
> When using an IDE, it might make more sense to use GDB/OpenOCD instead of stm8flash. stm8flash can only do basic stuff, such as read and write the flash and other memories. GDB/OpenOCD can do on-target debugging.

I found these instructions, in case anyone wants to see it, it looks like a good alternative, I'll try it:

https://github.com/hbendalibraham/stm8_started

rtek1000, 2023-04-09:
> I'm not sure I understand what you want stm8flash to do. You want it to automatically detect which model of MCU you have connected?

Now I see there is another option, I was searching the internet before I came to read the answers, and found the instructions already posted above. I was wondering about the debug options that the STM32CubeIDE has.

rtek1000, 2023-04-09:
Thanks all

closed 2023-04-09 by rtek1000