vdudouyt/stm8flash #174: Writing to Option Bytes not working

sbmarcom, 2024-01-14:
I could just be doing this wrong, but I am unable to write to the option byte area on my device.

When I run 

`./stm8flash -c stlink -p stm8s105c4 -s opt -r ../etc/test`

I correctly get the default option bytes as specified in the data sheet. 

However, when I try to write to them, then read again, nothing has changed. Here is my process:

I wrote a hex file as follows:

:02487E0055AA39
:00000001FF

The goal here is to write the boot loader enable bytes which are supposed be at 0x487E according to the data sheet.

Here is the command I ran to try and write this file:

`./stm8flash -c stlink -p stm8s105c4 -s opt -w ../ob.hex`

I have also tried the following, which also didn't work:

`./stm8flash -c stlink -p stm8s105c4 -s 0x487E -w ../ob.hex`

Please let me know if I am doing anything wrong, and if so, what I need to change to make this work. I read the main.c file, and it looks like specifying the address as either 'opt' or '0x487E' should work. 

Thanks,

Sam

joshuadavila, 2024-01-15:
try using an STVP generated hex file with bootloader enable flag active, it should be something like:

./stm8flash -c stlink -p stm8s105c4 -s opt -w ../ob.hex

and the contents of the file should be:

:0148000000B7
:0148010000B6
:0148030000B4
:0148050000B2
:0148070000B0
:0148090000AE
:01480B0000AC
:01480D0000AA
:01487E0055E4
:00000001FF

sbmarcom, 2024-01-15:
That didn't seem to work, when I read off the values at 0x487E it is still 0xFF00. 

I was reading the [programming manual page 22](https://www.st.com/resource/en/programming_manual/pm0051-how-to-program-stm8s-and-stm8a-flash-program-memory-and-data-eeprom-stmicroelectronics.pdf), and it seems to say that the FLASH_CR2 register must be set first: 

> Option bytes can be modified both in user/IAP mode and ICP mode (using the SWIM), except for the readout protection which can only be removed in ICP mode. In ICP mode, the OPT and NOPT bits of the FLASH_CR2 and FLASH_NCR2 registers must be set to their active state before attempting to modify the option bytes (OPT = 1 and NOPT = 0).

On page 23 it says that the FLASH_DUKR register must be set as well

>The option bytes are write protected (locked) until the correct MASS key is written in the FLASH_DUKR (with OPT set to ‘1’). It is possible to lock the memory again by resetting the DUL bit in the IAPSR register. If wrong keys are provided, another key program sequence can be performed without resetting the device.

Are these registers set by specifying `-s opt` in the command, or do I need to add this to the hex file as well? 

Thanks,

Sam

spth, 2025-10-06:
According to README, writing option bytes of stm8s105c4 with stlink is untested (but tested and working with stlinkv2).