vdudouyt/stm8flash #182: Cannot read STM8 flash using SWIM with stlinkv2 clones

abliss, 2025-07-16:
I am trying to read the flash of an stm8s105k6 using an stlinkv2 clone. I have had no success so far. After enabling debug output, I get this:
```
sudo ./stm8flash -p stm8s105k6 -c stlinkv2 -s flash -r fw00.bin
Determine FLASH area
     GET_VERSION
STLink: v2, JTAG: v37, SWIM: v7, VID: 8304, PID: 4837
     GET_CURRENT_MODE
        -> 03 00
     SWIM READBUFSIZE
        -> 0x1800
     SWIM READ_CAP 01
        -> 00 01 02 07 00 00 00 00
     SWIM ASSERT_RESET
     SWIM READSTATUS
        status 00 00 00 00
     SWIM ENTER_SEQ
     SWIM READSTATUS
        status 01 00 00 00
     SWIM READSTATUS
        status 01 00 00 00
     SWIM READSTATUS
        status 04 00 00 00
     SWIM READSTATUS
        status 04 00 00 00
     SWIM READSTATUS
        status 04 00 00 00
     SWIM READSTATUS
        status 04 00 00 00
     SWIM READSTATUS
        status 04 00 00 00
SWIM error 0x04
```

I have tried two seemingly-different clones at several different firmwares, and always the same result. I attached an oscilloscope to the SWIM line, and I always see a waveform like this:

[image]

the SWIM line starts high, then goes low for 20ms, then goes back high, and that's it. I was expecting to see the SWIM initialization sequence as described in the STMicro manual:

[image]

When I plug the stlinkv2 into a windows machine and run stlink's Visual Programmer, I do indeed see a waveform that looks like the SWIM initialization sequence:

[image]

Any idea what I'm doing wrong?

tdfocus-png, 2025-10-02:
I got more info but I am not really convinced there is much stm8flash can do to make this easier. 

I was able to get stm8flash to access my devboard if I first accessed it via STVP (in my case in a virtualbox Windows VM) and then disconnecting the power to my devboard and then reconnecting power.  Then I go to stm8flash and issue the -u and things work.

Now I've tried to do this power on/off trick just using stm8flash and it didn't seem to work for me.  So I am not sure if there is some state STVP sets the STLINK in that allows stm8flash to get the status ok from ENTER_SEQ.

tdfocus-png, 2025-10-03:
A correction to my last comment.  It ends up that if I power off my dev board I have to go through the dance with svtp before stm8flash can recognize my dev board through stlinkv2 and get past the 0x4 readstatus.  It seem like SVTP is doing something with the stlinkv2 that get stlinkv2 and my dev board able to talk to each other.

DeqingSun, 2025-10-29:
Just got a metal case one from Taobao. The main chip has no mark but the SWIM does work with stm8flash.
I believe some manufacturers did cost down design with a plastic case, and some changes in circuit.

[image]